igrr/libnmea-esp32 #1: Problem using libnmea-esp32 as component.

woiptcs, 2020-05-24:
Hi dear , 
I needed a NMEA parser for an ESP32 programed using ESP-IDF , that's why I arrived to your repo.
I succesfully compiled your example , after applying a patch .
[patch.txt](https://github.com/igrr/libnmea-esp32/files/4672786/patch.txt)
I'm having problems to include your code in another project.-
I tried allready 1000 combinations of Makefile and CMakeLists.txt without success and I'm becomming crazy .-

My project structure is : 
main -> src
components -> libnmea-esp32->libnmea

/home/noid/esp/AIDomotics/src/nas_ring8/main/nas_core.c:34:10: fatal error: libnmea.h: No such file or directory
 #include <libnmea.h>
          ^~~~~~~~~~~
compilation terminated.
make[1]: *** [/home/noid/esp/esp-idf/make/component_wrapper.mk:292: nas_core.o] Error 1
make: *** [/home/noid/esp/esp-idf/make/project.mk:610: component-main-build] Error 2

Please help!

woiptcs, 2020-05-24:
Additionally,
I see in libnmea-esp32 completelly absence of CMakeLists.txt , so I sopose I cannot use idf.py to compile and I need to use "make" instead , am I right ?

Thank you .-

woiptcs, 2020-06-04:
Fixed!

clement1993, 2020-10-12:
Hi any detailled of what you did in order to use it as default component ?

Thanks

woiptcs, 2020-10-18:
Finally I ended up using make , instad of Espresiff idf.py, cause idf.py does not support multiple levels of recursion , when including as default component.- 

Or that's what worked for me .-

igrr, 2021-01-27:
Have opened #3 to add CMake support. The warnings mentioned here will need to be fixed upstream.

closed 2021-02-01 by igrr