igrr/libnmea-esp32 #2: Compiling libnmea-esp32 with Cmake

JuhToiva, 2021-01-27:
Hi

Is there any change to get CmakeLists.txt file for this component ?

We are changing from gcc make to Cmake in our project and all other components are now working OK except this one.
I have been trying to make the change by myself but not that good with Cmake.

BR Juha T.

igrr, 2021-01-27:
Hi Juha, indeed the symbol renaming that is done in component.mk is a bit tricky to do in CMake. I'll try to come up with an approach to make it work.

igrr, 2021-01-27:
@JuhToiva Could you please try https://github.com/igrr/libnmea-esp32/pull/3?
Please let me know if that works. Unfortunately I don't have a GPS module at hand now, so I can't verify myself.

JuhToiva, 2021-01-29:
> Lähetetty: keskiviikko 27. tammikuuta 2021 20.05
> Vastaanottaja: igrr/libnmea-esp32 <libnmea-esp32@noreply.github.com>
> Kopio: Juha Toivanen <juha.toivanen@sadeinnovations.com>; Mention <mention@noreply.github.com>

Hello Ivan

Thanks for the fast answer.

Unfortunately, this does not work on my environment, I end up to linking problem:
/home/juhtoiva/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/libnmea-esp32/liblibnmea-esp32.a(parser_static.c.obj):(.literal.nmea_load_parsers+0x3c): undefined reference to `nmea_gpgsv_allocate_data'
/home/juhtoiva/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/libnmea-esp32/liblibnmea-esp32.a(parser_static.c.obj):(.literal.nmea_load_parsers+0x40): undefined reference to `nmea_gpgsv_set_default'
/home/juhtoiva/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/libnmea-esp32/liblibnmea-esp32.a(parser_static.c.obj):(.literal.nmea_load_parsers+0x44): undefined reference to `nmea_gpgsv_free_data'
/home/juhtoiva/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/libnmea-esp32/liblibnmea-esp32.a(parser_static.c.obj):(.literal.nmea_load_parsers+0x48): undefined reference to `nmea_gpgsv_parse'
/home/juhtoiva/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/libnmea-esp32/liblibnmea-esp32.a(parser_static.c.obj):(.literal.nmea_load_parsers+0x50): undefined reference to `nmea_gpgsv_init'
/home/juhtoiva/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/libnmea-esp32/liblibnmea-esp32.a(parser_static.c.obj): in function `nmea_load_parsers':
/home/juhtoiva/work/iot-asset-esp_copy_cmake/build/../components/libnmea-esp32/libnmea/src/nmea/parser_static.c:63: undefined reference to `nmea_gpgsv_init'
collect2: error: ld returned 1 exit status

BR Juha T.

igrr, 2021-01-29:
Could you please run `idf.py fullclean` and then `idf.py -v build`, then attach the full build output (it is going to be very long) as a file here? Also, could you tell which ESP-IDF version you are working with?

JuhToiva, 2021-01-29:
Hello

The ESP what is is in use is ESP32-WROVER-B and ESP IDF is v4.1.
(git clone -b v4.1 --recursive https://github.com/espressif/esp-idf.git esp_4/esp-idf)

In attachment is zipped build log.
Thanks for helping me.

Have a nice weekend Juha T.

Lähetetty: perjantai 29. tammikuuta 2021 15.28

igrr, 2021-01-29:
I'm sorry, I think Github doesn't accept the attachments over email. You'll need to open https://github.com/igrr/libnmea-esp32/issues/2 in the web browser and attach the archive to your comment... Alternatively, you can send it to me (ivan at *company name* .com)

igrr, 2021-01-29:
I wasn't able to reproduce the issue with IDF v4.1 tag and the bundled example (`example/` folder of the repository). The application linked successfully. If you are getting the issue with your custom application, could you also try building the provided example and see if the issue happens there as well?

JuhToiva, 2021-02-01:
Hello Ivan

When looking the Cmake log I did figure out the problem with linking in our environment.
There was one small change in code to remove the gpgsv parcer from normal build and add it only for debug compiling.
When removing that change linking was OK.

I did need to add also gpgsv to make:
set(parsers gpgga gpgll gprmc gpgsv)
as our debug code is using the gpgsv sentence also.

Thanks a lot for you great help.

Have a nice week Juha T.

Lähetetty: perjantai 29. tammikuuta 2021 19.58

igrr, 2021-02-01:
Thanks for testing this again, good to know it works for you now.
I didn't realize there was also a gpgsv parser, I'll update #3 to include it.

igrr, 2021-02-01:
Ah, i see now — GPGSV parser has been added in the following libnmea commit: https://github.com/jacketizer/libnmea/commit/adb7a210f4fbffaef99525b245b3a6f331304373. But this component is referencing an older libnmea commit: https://github.com/jacketizer/libnmea/commit/8122764b051aa0738b279791d396572179cc1f4e, which doesn't contain GPGSV. I will merge #3 first, and then update libnmea to the most recent commit, and include the new parser.

PR #3 (2021-02-01): add CMake build system support
This PR adds ESP-IDF CMake build system support.

Note that latest upstream libnmea library comes with CMakeLists.txt file, so theoretically we could simplify the component CMakeLists.txt file to set the correct options/flags and call `add_subdirectory`. I'll do this next time I update libnmea submodule to the latest upstream version.

Currently there are 2 warnings in libnmea submodule:

* stringop-truncation warning due to the use of strncpy in `NMEA_PARSER_PREFIX` macro. The compiler is correct, but the warning is actually harmless. The proper fix would be to  [... cut]
closed 2021-02-01 by igrr