igrr/libnmea-esp32 #4: Missing headers files nmea.h, gpgll.h, gpgga.h and gprmc.h

Yoimer, 2021-03-02:
Hello there @igrr. I hope you're doing fine. I just clone this repo and when checking its source code I didn't find the headers `nmea.h`, `gpgll.h`, `gpgga.h` and `gprmc.h`. The `nmea_example_main.c` file needs those to work.

I am attaching snapshot for clarifying.

Is there any way you can help me on this? I really want to use your library.

Thanks in advance.

[image]

igrr, 2021-03-02:
Can you try running `git submodule update --init` to fetch the contents of `libnmea` submodule?

Yoimer, 2021-03-02:
@igrr Thanks for responding Let me try that one.

Yoimer, 2021-03-03:
@igrr Hi. I just did what you suggested, and it worked. I got all the libraries now. Another question.

How could I use your code in my environment in order to test the GPS on the ESP32?

I am using `ESP-IDF 4.2`. However, I'm trying [UncleRus](https://github.com/UncleRus/esp-idf-lib) libraries to test a few sensors. Those libraries are not saved directly into `C:\esp-idf-v4.2\components` but in `C:\esp-idf-v4.2\esp-idf-lib\components`.
So for example, when I tried the `bmp180` example, I had to adapt the `CMakeLists.txt`, just like this:
I am passing snapshots for clarification. Looking forward to hearing your comments. Thanks a in advance 

**CMakeLists.txt**
`
cmake_minimum_required(VERSION 3.5)

set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/esp-idf-lib/components)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(example-bmp180)
`
[image]

**Code with all the libraries**
[image]

**IDF-4.2 Enviroment**
[image]

Yoimer, 2021-03-03:
@igrr Hello. I just figured out by myself and I could compile and flash your great library. I am using the`ESP32-WROOM-32D` and the `uBlox NEO-6M GPS`. I haven't wired it yet. But does this library work OK on these ones?. Thanks a lot

Yoimer, 2021-03-15:
@igrr I hope you're doing great. Have you ever tried this amazing code on AWS FreeRTOS ESP32? Thanks a lot

igrr, 2021-03-15:
No, I haven't tried in in AFR, only in esp-idf. That said, there is not much to libnmea-esp32 — it is only a makefile/CMakeLists.txt wrapper around libnmea library. Since all IDF features are available in AFR on the ESP32, I don't see a reason why it wouldn't work there, after adapting to the AFR build system.

Yoimer, 2021-03-15:
@igrr Thanks for replying. The thing is I am not well verse on AFR. I am struggling to import your (and many other libraries) to it.

closed 2021-05-19 by igrr