igrr/libnmea-esp32 #5: Fatal error: dlfcn.h.

Yoimer, 2021-03-25:
Hello there. It's me again. I am trying to compile your library on AFR V3.3 but it fails when compiling. I've been able to handle two sensors already on AFR, and want to use your gps library as well. 
 Could you be kind enough to help me out on this too, please?. Thanks in advance.
```
C:\FreeRTOS>cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -GNinja -S . -B demos/ota/build
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/ESP-IDF-Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Found Git: C:/git/cmd/git.exe (found version "2.31.0.windows.1")
-- Checking Python dependencies...
Python requirements from C:/FreeRTOS/vendors/espressif/esp-idf\requirements.txt are satisfied.
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
-- Component names: soc log heap freertos vfs newlib esp_ringbuf driver tcpip_adapter esp_event partition_table app_update spi_flash mbedtls micro-ecc
 bootloader_support efuse xtensa-debug-module app_trace ethernet nvs_flash pthread smartconfig_ack wpa_supplicant espcoredump esp32 cxx amazon-freerto
s-common dht esp_idf_lib_helpers i2cdev bmp180 gps secure_sockets freertos_plus_tcp bootloader nimble bt console esp_adc_cal esptool_py expat wear_lev
elling sdmmc fatfs freemodbus lwip nghttp openssl spiffs ulp
-- Component paths: C:/FreeRTOS/vendors/espressif/esp-idf/components/soc;C:/FreeRTOS/vendors/espressif/esp-idf/components/log;C:/FreeRTOS/vendors/espr
essif/esp-idf/components/heap;C:/FreeRTOS/vendors/espressif/boards/esp32/components/freertos;C:/FreeRTOS/vendors/espressif/esp-idf/components/vfs;C:/F
reeRTOS/vendors/espressif/esp-idf/components/newlib;C:/FreeRTOS/vendors/espressif/esp-idf/components/esp_ringbuf;C:/FreeRTOS/vendors/espressif/esp-idf
/components/driver;C:/FreeRTOS/vendors/espressif/boards/esp32/components/freertos_tcpip/tcpip_adapter;C:/FreeRTOS/vendors/espressif/esp-idf/components
/esp_event;C:/FreeRTOS/vendors/espressif/esp-idf/components/partition_table;C:/FreeRTOS/vendors/espressif/esp-idf/components/app_update;C:/FreeRTOS/ve
ndors/espressif/esp-idf/components/spi_flash;C:/FreeRTOS/vendors/espressif/boards/esp32/components/mbedtls;C:/FreeRTOS/vendors/espressif/esp-idf/compo
nents/micro-ecc;C:/FreeRTOS/vendors/espressif/esp-idf/components/bootloader_support;C:/FreeRTOS/vendors/espressif/esp-idf/components/efuse;C:/FreeRTOS
/vendors/espressif/esp-idf/components/xtensa-debug-module;C:/FreeRTOS/vendors/espressif/esp-idf/components/app_trace;C:/FreeRTOS/vendors/espressif/boa
rds/esp32/components/freertos_tcpip/ethernet;C:/FreeRTOS/vendors/espressif/esp-idf/components/nvs_flash;C:/FreeRTOS/vendors/espressif/esp-idf/componen
ts/pthread;C:/FreeRTOS/vendors/espressif/boards/esp32/components/freertos_tcpip/smartconfig_ack;C:/FreeRTOS/vendors/espressif/esp-idf/components/wpa_s
upplicant;C:/FreeRTOS/vendors/espressif/esp-idf/components/espcoredump;C:/FreeRTOS/vendors/espressif/esp-idf/components/esp32;C:/FreeRTOS/vendors/espr
essif/esp-idf/components/cxx;C:/FreeRTOS/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/amazon-freertos-common;C:/FreeRTOS/v
endors/espressif/boards/esp32/components/dht;C:/FreeRTOS/vendors/espressif/boards/esp32/components/esp_idf_lib_helpers;C:/FreeRTOS/vendors/espressif/b
oards/esp32/components/i2cdev;C:/FreeRTOS/vendors/espressif/boards/esp32/components/bmp180;C:/FreeRTOS/vendors/espressif/boards/esp32/components/gps;C
:/FreeRTOS/vendors/espressif/boards/esp32/components/secure_sockets;C:/FreeRTOS/vendors/espressif/boards/esp32/components/freertos_tcpip/freertos_plus
_tcp;C:/FreeRTOS/vendors/espressif/esp-idf/components/bootloader;C:/FreeRTOS/vendors/espressif/esp-idf/components/nimble;C:/FreeRTOS/vendors/espressif
/esp-idf/components/bt;C:/FreeRTOS/vendors/espressif/esp-idf/components/console;C:/FreeRTOS/vendors/espressif/esp-idf/components/esp_adc_cal;C:/FreeRT
OS/vendors/espressif/esp-idf/components/esptool_py;C:/FreeRTOS/vendors/espressif/esp-idf/components/expat;C:/FreeRTOS/vendors/espressif/esp-idf/compon
ents/wear_levelling;C:/FreeRTOS/vendors/espressif/esp-idf/components/sdmmc;C:/FreeRTOS/vendors/espressif/esp-idf/components/fatfs;C:/FreeRTOS/vendors/
espressif/esp-idf/components/freemodbus;C:/FreeRTOS/vendors/espressif/esp-idf/components/lwip;C:/FreeRTOS/vendors/espressif/esp-idf/components/nghttp;
C:/FreeRTOS/vendors/espressif/esp-idf/components/openssl;C:/FreeRTOS/vendors/espressif/esp-idf/components/spiffs;C:/FreeRTOS/vendors/espressif/esp-idf
/components/ulp
Loading defaults file C:/FreeRTOS/demos/ota/build/sdkconfig.defaults...
-- IDF_VER: v3.3-163-g601a03e
fatal: not a git repository (or any of the parent directories): .git
-- Project is not inside a git repository,                         will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Adding linker script C:/FreeRTOS/demos/ota/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/FreeRTOS/vendors/espressif/esp-idf/components/esp32/ld/esp32.rom.ld
-- Adding linker script C:/FreeRTOS/vendors/espressif/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/FreeRTOS/vendors/espressif/esp-idf/components/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/FreeRTOS/vendors/espressif/esp-idf/components/esp32/ld/esp32.rom.spiram_incompatible_fns.ld
-- Component libraries:
=========================Resolving dependencies==========================
                          All modules available

====================Configuration for FreeRTOS====================
  Version:                 202002.00
  Git version:             Unknown

Target microcontroller:
  vendor:                  Espressif
  board:                   ESP32-DevKitC
  description:             Development board produced by Espressif that comes in two
                           variants either with ESP-WROOM-32 or ESP32-WROVER module
  family:                  ESP32
  data ram size:           520KB
  program memory size:     4MB

Host platform:
  OS:                      Windows-6.1.7601
  Toolchain:               xtensa-esp32
  Toolchain path:          C:/Espressif/ESP-IDF-Tools/tools
[... 372 lines cut ...]
espressif/esp-idf/components/soc/include -mlongcalls -mlongcalls -Og -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -
Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-un
used-parameter -Wno-sign-compare -ggdb -std=gnu99 -Wno-old-style-declaration -MD -MT esp-idf/gps/CMakeFiles/idf_component_gps.dir/src/parser_static.c.
obj -MF esp-idf\gps\CMakeFiles\idf_component_gps.dir\src\parser_static.c.obj.d -o esp-idf/gps/CMakeFiles/idf_component_gps.dir/src/parser_static.c.obj
 -c ../../../vendors/espressif/boards/esp32/components/gps/src/parser_static.c
../../../vendors/espressif/boards/esp32/components/gps/src/parser_static.c:33:30: error: 'PARSER_COUNT' undeclared here (not in a function)
 nmea_parser_module_s parsers[PARSER_COUNT];
                              ^
../../../vendors/espressif/boards/esp32/components/gps/src/parser_static.c: In function 'nmea_load_parsers':
../../../vendors/espressif/boards/esp32/components/gps/src/parser_static.c:46:24: warning: unused variable 'parser' [-Wunused-variable]
  nmea_parser_module_s *parser;
                        ^
../../../vendors/espressif/boards/esp32/components/gps/src/parser_static.c:45:6: warning: unused variable 'i' [-Wunused-variable]
  int i = 0;
      ^
../../../vendors/espressif/boards/esp32/components/gps/src/parser_static.c:59:1: error: control reaches end of non-void function [-Werror=return-type]

 }
 ^
cc1.exe: some warnings being treated as errors
[581/799] Building C object esp-idf/nimble/CMakeFiles/idf_component_nimble.dir/nimble/nimble/host/util/src/addr.c.obj
ninja: build stopped: subcommand failed.

C:\FreeRTOS>
```

igrr, 2021-03-26:
> vendors/espressif/boards/esp32/components/gps/src/parser_static.c

I'm sorry, how did parser_static.c end up in the src directory directly under the component directory? Which CMakeLists.txt file are you using, the one in this repository, or the one in the upstream libnmea project? If you were using this project, the directory structure should have been components/<component name>/libnmea/src/parser_static.c.

Yoimer, 2021-03-26:
@igrr Hello and thanks a lot for your support. That's very kind from you. Please let me explain very detaily why `parser_static.c` end up in the src directory 

Basically the way I found to incorporate sensors to AFR and make them fully functional is as shown here, I tried the same with your library but it did not work:
https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/tree/master/vendors/espressif/boards/esp32/components

The sensors are `dht`, `bmp180` and their dependencies `i2cdev` , `esp_idf_lib_helpers`. Each folder has its own `include`, `src` and `CMakelists.txt` file.

I add these sensors as here https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/commit/61f6cbd3623d38d34492f30604dcebe745f6c880

In order to use those sensors, I created an extra task `sensors_task.c` which its header files is here [sensor.h](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/demos/include/sensors.hl) and  
 `src` here [sensors_task.c](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/demos/ota/sensors_task.c)
 
I declare the task here https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/0e8cdc3568095adc21f15d2148da666a821373d9/demos/CMakeLists.txt#L26 

and call it here 
https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/0e8cdc3568095adc21f15d2148da666a821373d9/vendors/espressif/boards/esp32/aws_demos/application_code/main.c#L141 all works fine.

So I tried to do the same with your library. I do not use your `CMakeLists.txt`. Instead I placed all your `headers.h` (include) file here https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/tree/master/vendors/espressif/boards/esp32/components/gps/include

and all the  `src` (source) here https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/tree/master/vendors/espressif/boards/esp32/components/gps/src

The `gps` has this `CMakeLists.txt` file https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/vendors/espressif/boards/esp32/components/gps/CMakeLists.txt

So would you please tell me what did I did wrong? How would you accomplish this using your very own l`ibnmea-esp32/libnmea/src/parser_static.c` ?

Thanks in advance @igrr

igrr, 2021-03-26:
> So would you please tell me what did I did wrong?

> I do not use your CMakeLists.txt. Instead I placed all your headers.h (include) file here [...]

Please see the [component CMakeLists.txt file](https://github.com/igrr/libnmea-esp32/blob/master/CMakeLists.txt) — there is a non-trivial amount of logic in there which you are omitting. By not including this code, the source files are getting compiled without the necessary compilation flags defined, leading to the errors you are seeing. The [component CMakeLists.txt file you are using](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/vendors/espressif/boards/esp32/components/gps/CMakeLists.txt) doesn't contain the equivalent logic.

It seems that you are unable to use the component CMakeLists.txt file provided in this repository because it doesn't support ESP-IDF 3.3. The thing is, in ESP-IDF 3.3, different syntax is used in CMakeLists.txt files (`register_component` instead of `idf_component_register`). I'll see about making the CMakeLists.txt file in this component compatible with IDF 3.3.

In the meantime you can try to adapt the logic from [this component CMakeLists.txt file](https://github.com/igrr/libnmea-esp32/blob/master/CMakeLists.txt) to [yours](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/vendors/espressif/boards/esp32/components/gps/CMakeLists.txt).

Yoimer, 2021-03-26:
@igrr Thanks for responding. I will try adapting the logic. Thanks a lot again

igrr, 2021-03-31:
https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/vendors/espressif/boards/esp32/components/gps/CMakeLists.txt looks the same as it was before, can you point to your changes and the errors you got?

Yoimer, 2021-05-05:
@igrr Hello. I hope you're doing great. I am retaking this, for AWS FreeRTOS ESP32 ESP-IDF 3.3. Would you be kind enough to guide me on how to use your [component CMakeLists.txt file ](https://github.com/igrr/libnmea-esp32/blob/master/CMakeLists.txt) on `ESP-IDF 3.3`, please?
I have no experience on CMake. Your library is the most reliable I've tested so far. Thanks in advance.

igrr, 2021-05-19:
Hi @Yoimer, IDF v3.3 should be supported now in the main branch.

closed 2021-05-19 by igrr

Yoimer, 2021-05-20:
> Hi @Yoimer, IDF v3.3 should be supported now in the main branch.

@igrr Hello. Thanks a lot for taking your time and supporting the v3.3 on this amazing library. I will test it and let you know.