igrr/libnmea-esp32 #6: How to validate when GPS is physically attached to ESP32

Yoimer, 2021-05-26:
Hello there @igrr. Hope you are doing great.

I am testing the latest release of the library and I found myself with an issue. 

When the GPS is not physically attached to the ESP32 it simply does nothing. I guess it is because UART on ESP32 does not receive any electrical signal from the GPS.

Is there any way (a conditional for example) that I could use in order to validate if the GPS is attached to board?

Like, if gps attached carry on with the rest of the code, but it is not, save a `-200` in a flag (just to say a number) and do anything else.

Looking forward to hearing your comments.

Thanks in advace.

igrr, 2021-05-30:
Hi @Yoimer, please check this part of the example:

https://github.com/igrr/libnmea-esp32/blob/32858b4a50107d90c43d63ca98900b5c5ec27333/example/main/nmea_example_main.c#L57-L64

You can adjust the timeout for reading the data and change the behavior if no data is received from UART.

Yoimer, 2021-05-30:
> Hi @Yoimer, please check this part of the example:
> 
> https://github.com/igrr/libnmea-esp32/blob/32858b4a50107d90c43d63ca98900b5c5ec27333/example/main/nmea_example_main.c#L57-L64
> 
> You can adjust the timeout for reading the data and change the behavior if no data is received from UART.

@igrr Thanks a lot for replying. I will test this and let you know.

Yoimer, 2021-06-02:
@igrr I am closing this. Now I am able to validate whether UART has received data or not. Thanks a lot for your support.

closed 2021-06-02 by Yoimer