igrr/libnmea-esp32 #9: Integration with AFR ESP32 V202007

Yoimer, 2021-05-31:
@igrr Hello. I hope you're doing great.

I want to use your library  as a component on `AWS FreeRTOS ESP32 V202007` using `ESP-IDF V3.3`.

I was able to incorporate the `minmea`  library on AFR as referenced here : [AFR-CMakeLists](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007/blob/4eef8f68f87a8ff2e41abc73127f4a35ebe0824d/vendors/espressif/boards/esp32/CMakeLists.txt#L477)

The implementation (which includes a JSON format for AWS IoT) is here [sensors_task.c](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007/blob/integration/demos/ota/sensors_task.c). It works fine, but I want to test yours as well.
 
What really confuses me is how to add on AFR this [CMakeLists.txt](https://github.com/igrr/libnmea-esp32/blob/main/CMakeLists.txt) from `libnmea-esp32 `  on [AFR-CMakeLists.txt.](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007/blob/4eef8f68f87a8ff2e41abc73127f4a35ebe0824d/vendors/espressif/boards/esp32/CMakeLists.txt)

I am really eager to use what you commented here  https://github.com/igrr/libnmea-esp32/issues/6#issuecomment-850992299

regarding no receiving data on UART.

In [readLine](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007/blob/4eef8f68f87a8ff2e41abc73127f4a35ebe0824d/demos/ota/sensors_task.c#L45) `minmea`  I could not use something similar.

Your library is the most accurate when it comes to GPS for ESP32. 

Would you be kind enough to assist me on this, please?

I really appreciate your time for updating your amazing library for `ESP-IDF V3.3.

Looking forward to reading from you soon.

igrr, 2021-05-31:
> What really confuses me is how to add on AFR this CMakeLists.txt from libnmea-esp32 on AFR-CMakeLists.txt.

You don't :) You need to put this repository into a subdirectory of [`components`](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007/tree/4eef8f68f87a8ff2e41abc73127f4a35ebe0824d/vendors/espressif/boards/esp32/components), like you did with minmea.

> I am really eager to use what you commented here #6 (comment)
> regarding no receiving data on UART.
> In readLine minmea I could not use something similar.

I'd recommend you taking the example provided in this repository (https://github.com/igrr/libnmea-esp32/blob/main/example/main/nmea_example_main.c), and copying the `uart_setup` and `read_and_parse_nmea` functions (along with any includes they require) into your project. Once you get that working, you can add the AWS IoT stuff back in.

Yoimer, 2021-05-31:
> You don't :) You need to put this repository into a subdirectory of [...]

@igrr Thanks for replying. Definitely, I will try that out and let you know.

Yoimer, 2021-06-01:
@igrr :+1: Thanks a lot for your guidance. I did as you recommended and now I am able to use `libnmea-esp32` on `AFR202007`.

You might check the libraries here [libnmea-esp32-libraries](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007/blob/b2d7344f4d2b9ef824850fc800f16bac4f53d11c/vendors/espressif/boards/esp32/CMakeLists.txt#L486)

As a component [libnmea-esp32-as-a-component](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007/tree/b2d7344f4d2b9ef824850fc800f16bac4f53d11c/vendors/espressif/boards/esp32/components/libnmea-esp32) and here [CMakeLists.txt](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007/blob/b2d7344f4d2b9ef824850fc800f16bac4f53d11c/vendors/espressif/boards/esp32/CMakeLists.txt#L486)

Now I will try the AWS IoT section and let you know.

Thanks a lot again! :)

Yoimer, 2021-06-02:
@igrr Hello again. Hope you are doing great.

I tested the `libnmea-esp32` library here [sensors_task.c](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007/blob/e1b872b981a77f3ef977c870f73cfdfc49d6cbc8/demos/ota/sensors_task.c) and it runs fine. I modified just a bit (for my own app) but its core is exactly the same.

However, when the gps (`NEO-6M-0-001`) reads data from the satellites it is not that accurate. It shows latitude and longitude which is not even close where I tested. Please read this serial output [libnema_esp_32_serial.txt](https://github.com/igrr/libnmea-esp32/files/6581394/libnema_esp_32_serial.txt).

On the other hand, when I test the [minmea.h](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007/blob/4eef8f68f87a8ff2e41abc73127f4a35ebe0824d/vendors/espressif/boards/esp32/components/gps/include/minmea.h) and [minmea.c](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007/blob/4eef8f68f87a8ff2e41abc73127f4a35ebe0824d/vendors/espressif/boards/esp32/components/gps/src/minmea.c) library, implemented here [sensors_task-minmea](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007/blob/4eef8f68f87a8ff2e41abc73127f4a35ebe0824d/demos/ota/sensors_task.c) (using the same gps at the same location) it does shows me a very precise location. Please check the output.
[minmea_serial.txt](https://github.com/igrr/libnmea-esp32/files/6581431/minmea_serial.txt)
When I pass those coordinates on google maps or any service it shows me almost exactly where I am.

So my questions are:

- Is there any way I could get a more accurate location using `libnmea-esp32`, similar as in `minmea` ?

- Would you please assist me as usual on this?

I am quite fascinated with this implementation. It does have an excellent error checking when no UART activity present.

Looking forward to hearing your recommendations.

Thanks in advance,

Yoimer

igrr, 2021-06-02:
Could you please attach the raw NMEA message lines that are parsed
incorrectly?

For example, you can print the message received from UART to console before
passing it to nmea_parse.

Yoimer, 2021-06-02:
Hello @igrr. Thanks for your support. Yes, I am attaching the what the buffer has before the values are parsed. [libnema_esp_32_buffer.txt](https://github.com/igrr/libnmea-esp32/files/6584989/libnema_esp_32_buffer.txt)

I just commited the buffer printing as seen here [sensors_task.c](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007/blob/d9f2d71018b5435fdb98d38f3fdd7c1560c8f6f6/demos/ota/sensors_task.c#L65)

If this helps, the expected coordinates are `10.470182,-66.792351` or near of course

Looking forward to reading your comments and recommendations.

Thanks in advance.

igrr, 2021-06-02:
The output produced by libnmea looks pretty much spot on... Are you converting between decimal and degree/minutes/second representations correctly?

Let's look at this message:
```
$GPRMC,134545.00,A,1028.21100,N,06647.53643,W,0.185,,020621,,,A*66
```

libnmea output from your log was:
```
GPRMC sentence
Longitude:
  Degrees: 66
  Minutes: 47.536430
  Cardinal: W
Latitude:
  Degrees: 10
  Minutes: 28.211000
  Cardinal: N
```

66° 47.5364' = 66 + (47.5364) / 60.0 = 66.792273.
Your expected longitude is 66.792351, difference is 7.7e-5 degrees, or about 8.6 meters.

10° 28.211000' = 10 + (28.211000) / 60.0 = 10.470183.
Your expected latitude is 10.470182, difference is 1.3e-6 degrees, that's less than a meter.

Yoimer, 2021-06-02:
> The output produced by libnmea looks pretty much spot on... Are you converting between decimal and degree/minutes/second representations correctly?
>
> Let's look at this message:
>
> ``` [...]

@igrr Thanks a lot for replying. I did not know about dividing the minutes by 60 in order to get the accurate you have just shown. I will implement that and let you.
Once again, thanks a lot for your support on this.

igrr, 2021-06-02:
In that case I recommend you to check https://en.wikipedia.org/wiki/Geographic_coordinate_conversion#Change_of_units_and_format.

Yoimer, 2021-06-02:
> In that case I recommend you to check https://en.wikipedia.org/wiki/Geographic_coordinate_conversion#Change_of_units_and_format.
> libnmea uses the 2nd format mentioned, "degrees and decimal minutes". You need to convert either to decimal degrees (used in, for example, Google Maps coordinates), or into sexagesimal (used sometimes for human-readable representation).

@igrr Sure thing. Already working on this. Thanks a lot for your support.

Yoimer, 2021-06-03:
@igrr Hello, I hope you are doing great.

For the  West `W` and  South `S` coordinates, I needed to make this simple logic [longitude](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007/blob/7392ea37d99f906fecd2a863b4fc853f790962ed/demos/ota/sensors_task.c#L120 ) and [latitude](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007/blob/7392ea37d99f906fecd2a863b4fc853f790962ed/demos/ota/sensors_task.c#L141) to get the `negative` values of those.

So my questions are:

- Does `libnmea-esp32` library have any built-in method to get the negative coordinates when it has `W` and `S` or did I miss something else to get those?

- If not, would it be that difficult to implement that feature?

I am quite happy using your library. Almost ready to finish this amazing firmware implementation, and that's because of your help.

Looking forward to hearing from your comments and recommendations.

Respectfully,

Yoimer

igrr, 2021-08-12:
Hi @Yoimer, sorry I didn't reply before. Did you manage to get it working? The logic you have linked is fine, except that longitude cardinal can not be `S` and latitude can not be `W`.

I think in general libnmea could make use of some structures for representing coordinates, and functions to convert between them. However it's probably best to do it in the libnmea itself, not in this wrapper, since such conversions don't need to be ESP-IDF-specific.

Yoimer, 2021-08-12:
@igrr Thanks for responding. I managed it to have it as shown on the links. Please, if you can also help me to do it directly on libnmea itself, I'll really appreciate it.
Thanks in advance.

closed 2023-02-25 by igrr