igrr/libnmea-esp32 #15: I2C example

beriberikix, 2022-10-22:
Hi Ivan! Thanks for making this port to the new component system :)

Would it be possible for someone to create an I2C example? I'm interested in a specific use case to support the [PA1010D](https://www.adafruit.com/product/4415) which is used in the open source Ribbit Network project (see more [here](https://github.com/Ribbit-Network/ribbit-network-frog-sensor/issues/174).)

igrr, 2022-11-07:
I got the GPS in question and will try to add an example in the next couple of weeks.

igrr, 2022-11-13:
@beriberikix you are welcome to try #16.

I haven't had a chance to test if UART still works after these changes, but I2C seems to work.

akhilgupta1093, 2022-11-30:
Hi @igrr ! Thanks so much for this implementation. I'm trying it out for Ribbit Network, but it seems like the only data coming from the [PA1010D](https://www.adafruit.com/product/4415) GPS is in the form of GPGSA and GPGSV sentences, which don't contain position data ([reference](https://www.rfwireless-world.com/Terminology/GPS-sentences-or-NMEA-sentences.html)).

It seems like the device is sending other sentence types, although, weirdly, I can't find too much information about these other sentence types (GLGSA, GNRMC, GNVTG, GNGGA...)

You can see the code I'm using [here](https://github.com/akhilgupta1093/ribbit-network-frog-sensor/blob/gps_exploration/v4/software/main/app_main.c)

Did you manage to receive other sentence types from your PA1010D GPS? If so, do you have an idea about what I'm doing wrong?

akhilgupta1093, 2022-11-30:
Seems like GNRMC data may be what I want from the device ([GNRMC reference](https://navspark.mybigcommerce.com/content/NMEA_Format_v0.1.pdf)) -- I guess this isn't currently supported in the NMEA package. Do you have a suggestion for a workaround/another package to use?

igrr, 2022-11-30:
@akhilgupta1093 Do you specifically need G**N**RMC (i.e. GPS+GLONAS) and not G**P**RMC (GPS)?

If you want to configure the receiver to send particular sentences or enable/disable particular talkers (GPS/GPS+GLONAS) you need to send some commands to it. The commands are vendor-specific. Adafruit has made the command reference for this module available [here](https://cdn-shop.adafruit.com/datasheets/PMTK_A11.pdf). You can also check [their tutorial](https://learn.adafruit.com/adafruit-mini-gps-pa1010d-module/circuitpython-python-i2c-usage#echotest-example-3047465), it has a few sample PMTK commands.

igrr, 2022-11-30:
I've opened an upstream PR to add support for `GN*` sentences in libnmea: https://github.com/jacketizer/libnmea/pull/58.

akhilgupta1093, 2022-11-30:
Thanks for the references, I'll take a look at enabling/disabling particular talkers. We only need location data, not satellite data, so GP* would be good enough for us. In any case, I look forward to your PR being merged. Thanks a ton for your help!

akhilgupta1093, 2022-12-01:
@igrr I'm having a little trouble configuring the receiver like you mentioned above (the examples are in circuitpython, I'm using esp-idf and c). I've decided to wait for your `GN*` support to be merged into libnmea-esp32 as the solution to my issue. No time rush, of course, just let me know when this is done, thanks

akhilgupta1093, 2022-12-04:
@igrr I see that your [PR](https://github.com/jacketizer/libnmea/pull/58) has now been merged into the jacketizer/libnmea repo! If you update your fork of the repo, I think I should be good to go, if I'm not mistaken

igrr, 2022-12-05:
@akhilgupta1093 I have updated https://github.com/igrr/libnmea-esp32/pull/16, please give it a try.

igrr, 2022-12-06:
You can modify the idf_component.yml in your project to install the component from a git repository, until a new version is released:
```yml
dependencies:
  libnmea-esp32:
    git: https://github.com/igrr/libnmea-esp32.git
    version: feature/i2c
```
(or `version: <git commit id>`)

igrr, 2022-12-06:
Are you sure the module actually has a fix on the satellites?

I have put my module next to the window and it took about 5 minutes for the module to acquire a fix. (I have a relatively clear view of the sky outside; if you have other buildings nearby it may take even more time.)

akhilgupta1093, 2022-12-06:
Thanks for the tip, seems like there's data being ingested now, but for some reason it's facing an error trying to read the data bytes. (here's my [branch](https://github.com/Ribbit-Network/ribbit-network-frog-hardware/pull/189) for reference)

Getting this error log repeatedly:
```
E (26673) frog_sensor: Failed to read byte over I2C: ESP_FAIL (0xffffffff)
```

akhilgupta1093, 2022-12-07:
Ah it worked after disconnecting and re-connecting the wires. Thanks so much for all your help! Seems like the integration is complete on our part.

closed 2023-02-25 by igrr